euslisp/jskeus #480: ROS_PACKAGE_PATHにバックスラッシュが入る

mmurooka, 2018-01-10:
最近のeuslisp, jskeusを`catkin build`して`source setup.bash`でセットされるROS_PACKAGE_PATHにバックスラッシュが入るようになってしまっているようです．

```
source /opt/ros/indigo/setup.bash
cd /tmp/
mkdir -p catkin_ws_euslisp/src
cd catkin_ws_euslisp/src
wstool init
wget https://raw.githubusercontent.com/jsk-ros-pkg/jsk_roseus/master/setup_upstream.sh
bash setup_upstream.sh -w ..
cd ..
catkin build
echo $ROS_PACKAGE_PATH | sed "s/:/\n/g"
```
で
```
/tmp/catkin_ws_euslisp/src/jsk-ros-pkg/geneus
/tmp/catkin_ws_euslisp/src/euslisp/Euslisp
/tmp/catkin_ws_euslisp/src/euslisp/jskeus\
/tmp/catkin_ws_euslisp/devel/share/euslisp\
/tmp/catkin_ws_euslisp/devel/share/jskeus
/opt/ros/indigo/share
/opt/ros/indigo/stacks
```
というようになって，devel/share/euslispとdevel/share/jskeusの前に
バックスラッシュが入っています．

https://github.com/fkanehiro/hrpsys-base/issues/1112 と同様の問題で，
このcatkin workspaceをチェーンしたworkspaceでのコンパイル時に"File name too long"というエラーが出ます．

hrp2017用の新しいcalでこの問題が起きて気づきました． cc @iory

mmurooka, 2018-01-10:
euslispとjskeusはaptで良い，sourceから入れる手順が違う，とかも気にしないといけないポイントですね．

mmurooka, 2018-01-10:
https://github.com/fkanehiro/hrpsys-base/issues/1112 が https://github.com/fkanehiro/hrpsys-base/commit/e335e24f614d799214ae92d363e919456bbd0cbb で起き始めたので，
今回は https://github.com/tork-a/euslisp-release/commit/3aceb99899aafb98ac05d8f86d0f712b1ef5c8da と https://github.com/tork-a/jskeus-release/commit/ba596b040d82578b42fce05b95c98b921e8bc206 が原因になっていそうです．

k-okada, 2018-01-10:
> fkanehiro/hrpsys-base#1112
> <https://github.com/fkanehiro/hrpsys-base/issues/1112> が
> fkanehiro/hrpsys-base@e335e24
> で起き始めたので，
> 今回は tork-a/euslisp-release@3aceb99
> と tork-a/jskeus-release@ba596b0
> が原因になっていそうです．

なるほど．
https://github.com/tork-a/jskeus-release/pull/4
https://github.com/tork-a/euslisp-release/pull/12
のCMakeLists.txtを手元のものと入れ替えてもんだいないか確認してもらえると嬉しいです．

mmurooka, 2018-01-10:
ありがとうございます．

```bash
$ echo $ROS_PACKAGE_PATH | sed "s/:/\n/g"
/tmp/catkin_ws_euslisp/src/jsk-ros-pkg/geneus
/tmp/catkin_ws_euslisp/src/euslisp/Euslisp
/tmp/catkin_ws_euslisp/src/euslisp/jskeus
/opt/ros/indigo/share
/opt/ros/indigo/stacks
```
となって，問題のバックスラッシュはなくなりましたので，これで大丈夫そうです．
合わせてdevel/shareの下のパス自体がなくなりましたが，これはきっと本来なくて良いものですね．

k-okada, 2018-01-10:
あれ，それは意図していないかも．
https://github.com/tork-a/euslisp-release/blob/release/indigo/euslisp/CMakeLists.txt#L41
にあるようにcatkin bのときにeuslisp的にはmake installまでやってしまって，
catkin の中のcmake のinstallでは，このinstallされたソースツリーをコピーしています．
なので，python pathが devel側に来るのと同じで，eusのものもそちらにくる設定のつもりでした．
あまり使い勝手は良くないですが，src以下をいじってcatkin bしてdevelに持って行って，その先がroscdされると．
手順としては一応筋が通っているはずです．うーむ．どうしようか．

mmurooka, 2018-01-10:
なるほど．https://github.com/fkanehiro/hrpsys-base/issues/1112 , https://github.com/fkanehiro/hrpsys-base/pull/1114 でも，バックスラッシュがなくなってくれた代わりに，devel以下のパスが消えていました．（ということを当時のPRのコメントなどで共有できておりませんでした．．）