euslisp/jskeus #487: pointcloudの:point-listの値が動かない

shintarokkk, 2018-02-23:
pointcloudクラスの\*point\*を\*robot\*のリンクにassocして、\*robot\*が動いたらそのpointもついてくるようにしているのですが、その際に(send \*point\* :point-list)で返ってくる値が常に一定になってしまっています。
描画されている通りのpointの位置を取ってきたいのですが、どのようにすれば良いか教えていただけないでしょうか。

以下に例を貼ります。
;;setup                                                                                                                             
(load "package://hrpsys_ros_bridge_tutorials/euslisp/samplerobot-interface.l")
(setq \*robot\* (instance samplerobot-robot :init))
(setq \*point\* (instance pointcloud :init
                        :points (list #f(500 0 1000))
                        :point-size 100.0))
(send \*robot\* :assoc \*point\*)
(objects (list \*robot\* \*point\*))
[image]

(send \*point\* :point-list)
**→ (#f(500.0 0.0 1000.0))が返ってくる**

;;move robot and point                                                                                                              
(send \*robot\* :translate #f(0 0 200))
(send \*point\* :worldcoords)
(send \*irtviewer\* :draw-objects)
[image]

(send \*point\* :point-list)
**→ (#f(500.0 0.0 1000.0))が返ってくる
　(ここで#f(500.0 0.0 1200.0)が返ってきてほしい**

mmurooka, 2018-02-23:
```
(mapcar #'(lambda (p) (send (send *point* :parent) :transform-vector p)) (send *point* :point-list))
```
で取ってこれますが，専用のメソッドがあってもいいかもしれませんね．

shintarokkk, 2018-02-23:
@mmurooka さん
ありがとうございます。とりあえずこれで対応したいと思います。

parentの初期位置を保存しておいて、その動きに合わせてparrayの中身を書き換えるようなメソッドも試しに作ってみましたが、そうすると今度は描画の方が本来の位置とずれてしまう(parrayの中身を動かした分とassocの効果で動かされる分が二重に働いてしまっている?)という問題が起きてしまいました。

shintarokkk, 2018-02-23:
すみません、上に挙げた例が悪く、上記の方法のみでは解決していませんでした。
parentの初期位置を\*init-base\*に保存しておいて、
`
(mapcar #'(lambda (p) (send (send *init-base* :transformation (send *point* :parent)) :transform-vector p)) (send *point* :point-list))
`
のようにする必要がありました。(最初の例だと\*robot\*の初期位置が原点だったので問題無く見えていた)

以下のような例で確認できます

```
;;setup                                                                                                                             
(load "package://hrpsys_ros_bridge_tutorials/euslisp/samplerobot-interface.l")
(setq *robot* (instance samplerobot-robot :init))
(setq *point* (instance pointcloud :init
                        :points (list #f(500 0 1000))
                        :point-size 100.0))
(send (send *robot* :waist_link1_lk) :assoc *point*)
(setq *init-base* (send (send *robot* :waist_link1_lk) :copy-worldcoords))
(objects (list *robot* *point*))

;;at this point                                                                                                                     
(send *point* :point-list) ;returns (#f(500.0 0.0 1000.0))                                                                          
(mapcar #'(lambda (p) (send (send *point* :parent) :transform-vector p)) (send *point* :point-list)) ;returns (#f(500.0 0.0 1899.5)\
)                                                                                                                                   
(mapcar #'(lambda (p) (send (send *init-base* :transformation (send *point* :parent)) :transform-vector p)) (send *point* :point-li\
st)) ;returns (#f(500.0 0.0 1000.0))                                                                                                

;;move robot and point                                                                                                              
(send *robot* :translate #f(0 0 200))
(send *point* :worldcoords)
(send *irtviewer* :draw-objects)

;;at this point                                                                                                                     
(send *point* :point-list) ;returns (#f(500.0 0.0 1000.0))                                                                          
(mapcar #'(lambda (p) (send (send *point* :parent) :transform-vector p)) (send *point* :point-list)) ;returns (#f(500.0 0.0 2099.5)\
)                                                                                                                                   
(mapcar #'(lambda (p) (send (send *init-base* :transformation (send *point* :parent)) :transform-vector p)) (send *point* :point-list)) ;returns (#f(500.0 0.0 1200.0))
```

mmurooka, 2018-02-23:
```
(mapcar #'(lambda (p) (send (send *point* :worldcoords) :transform-vector p)) (send *point* :point-list))
```
でもだめかな．最初からこっちを書くほうが正しかった気がします．

shintarokkk, 2018-02-23:
@mmurooka さん、ありがとうございます。その方法でうまくいきました。

YoheiKakiuchi, 2018-02-26:
pointcloudクラスは、データ数が多いことを前提にしていて、計算量が多くなるので、
worldcoordsが変更されてもデータを変更しません。
viewer上の表示はworldcoordsに依存して変更されます。

irtviewer上で正しく見えていれば、
@murooka の方法で正しく変換できます。点数が少ない時はこれで問題ありません。

それと同じことをしてくれるメソッドが
```send *point* :convert-world``` です。これは、*point*のデータを書き換えます。
assocされていると、表示（やworldcoords）がおかしくなるかもしれません。
これでどうなるか確認してみてください。

@murooka の方法の高速版が以下です。
```(send (send *point* :convert-world :create t) :point-list)```
```:crete t```をつけると、```*point*```オブジェクトを変更しないで新しいインスタンスを返します。

shintarokkk, 2018-02-26:
@YoheiKakiuchi さん

pointcloudをどこかにassocしてconvert-to-worldを送ると、描画上で望ましくない移動が起こってしまいます。
例えば、(robotは上記の例と同様に準備して)

```
(setq *point* (instance pointcloud :init :points (list #f(1000 0 1000)) :point-size 100))
(objects (list *robot* *point*))
(send (send *robot* :waist_link1_lk) :assoc *point*))
(send *point* :convert-to-world)
```

としてから描画すると、点の位置がずれてしまいます。
\*point\*の中身は以下のようになっていました。
```
;;生成時
((plist) (rot . #2f((1.0 0.0 0.0) (0.0 1.0 0.0) (0.0 0.0 1.0))) (pos . #f(0.0 0.0 0.0)) (parent) (descendants) (worldcoords . #<coordinates #X5b9e208  0.0 0.0 0.0 / 0.0 0.0 0.0>) (manager . #<pointcloud #X5cc1cc8  0.0 0.0 0.0 / 0.0 0.0 0.0>) (changed) (parray . #2\
f((1000.0 0.0 1000.0))) (carray) (narray) (curvature) (pcolor . #f(0.0 1.0 0.0)) (psize . 100) (awidth . 2.0) (asize . 0.0) (box . #<bounding-box #X62cc028 #f(1000.0 0.0 1000.0)/#f(1000.0 0.0 1000.0)>) (height . 1) (width . 1) (view-coords . #<coordinates #X5b9c75\
0  0.0 0.0 0.0 / 0.0 0.0 0.0>) (drawnormalmode . :normal) (transparent) (tcarray))
;;assocしたあと
((plist) (rot . #2f((1.0 0.0 0.0) (0.0 1.0 0.0) (0.0 0.0 1.0))) (pos . #f(0.0 0.0 -899.5)) (parent . #<bodyset-link #X5d9e1e0 WAIST_LINK1  0.0 0.0 899.5 / 0.0 0.0 0.0>) (descendants) (worldcoords . #<coordinates #X5b9e208  0.0 0.0 0.0 / 0.0 0.0 0.0>) (manager . #<\
pointcloud #X5cc1cc8  0.0 0.0 0.0 / 0.0 0.0 0.0>) (changed) (parray . #2f((1000.0 0.0 1000.0))) (carray) (narray) (curvature) (pcolor . #f(0.0 1.0 0.0)) (psize . 100) (awidth . 2.0) (asize . 0.0) (box . #<bounding-box #X62cc028 #f(1000.0 0.0 1000.0)/#f(1000.0 0.0 \
1000.0)>) (height . 1) (width . 1) (view-coords . #<coordinates #X5b9c750  0.0 0.0 0.0 / 0.0 0.0 0.0>) (drawnormalmode . :normal) (transparent) (tcarray))
;;convert-to-worldしたあと
((plist) (rot . #2f((1.0 0.0 0.0) (0.0 1.0 0.0) (0.0 0.0 1.0))) (pos . #f(0.0 0.0 0.0)) (parent . #<bodyset-link #X5d9e1e0 WAIST_LINK1  0.0 0.0 899.5 / 0.0 0.0 0.0>) (descendants) (worldcoords . #<coordinates #X5b9e208  0.0 0.0 899.5 / 0.0 0.0 0.0>) (manager . #<p\
ointcloud #X5cc1cc8  0.0 0.0 899.5 / 0.0 0.0 0.0>) (changed) (parray . #2f((1000.0 0.0 1000.0))) (carray) (narray) (curvature) (pcolor . #f(0.0 1.0 0.0)) (psize . 100) (awidth . 2.0) (asize . 0.0) (box . #<bounding-box #X62cc028 #f(1000.0 0.0 1000.0)/#f(1000.0 0.0\
 1000.0)>) (height . 1) (width . 1) (view-coords . #<coordinates #X5b9c750  0.0 0.0 0.0 / 0.0 0.0 0.0>) (drawnormalmode . :normal) (transparent) (tcarray))
```
convert-to-worldによってposが変わってしまうようです。

closed 2018-02-27 by k-okada