euslisp/jskeus #514: Whether the target-coords argument of inverse-kinematics function should be able to take a function.

iory, 2018-07-19:
Related to #503 

I do not know the use case of target-coords as function.
Someone use target-coords as a function?
If you use target-coords as a function, Please write the use-case to this thread.

inverse-kinematicsの引数のtarget-coordsを関数として使っている人はいるでしょうか？
いたら教えてください。

furushchev, 2018-07-19:
初めて知ったほどで，とくに使っていませんでした．
euslibとjsk-ros-pkgの下で`ag -r ":inverse-kinematics *#'"や`ag -r ":inverse-kinematics *f"`などでゆるく検索してみましたが，使用例は見当たりませんでした．
by ライトユーザ

mmurooka, 2018-07-30:
https://github.com/euslisp/jskeus/blob/master/irteus/demo/look-at-ik.l#L23-L38
に使用例があります．

以下の例のように，
左手は，`:translation-axis :z :rotation-axis nil`にして，右手を（左手ローカル座標系で）左手の100mm上に持ってきたい場合では，関数をサポートしないと上手く扱えないと思います．ここでは，右手のend-coordsに応じたcoordinateを返す関数を，左手のmove-targetとして指定しています．
[image]
```
(load "irteus/demo/sample-robot-model.l")

(defun test-inverse-kinematics-target-coords-function-with-dual-arm
  ()
  "
   Solve IK to put the right hand 100 mm above the left hand.
  "
  (unless (boundp '*robot*)
    (setq *robot* (instance sample-robot :init)))
  (setq *obj* (make-cylinder 20 100))
  (send *obj* :set-color #f(1 1 0))
  (send *robot* :reset-pose)
  (objects (list *robot* *obj*))

  (send *robot* :inverse-kinematics
        (list (make-coords :pos #f(400 0 0))
              )
        :move-target
        (list (send *robot* :larm :end-coords)
              )
        :link-list
        (list (send *robot* :link-list
                    (send (send *robot* :larm :end-coords) :parent)
                    (car (send *robot* :larm :links))
                    )
              )
        :translation-axis (list t)
        :rotation-axis (list nil)
        )

  (let* ((cnt 0.0)
         )
    (do-until-key
     (incf cnt 0.1)
     (send *robot* :inverse-kinematics
           (list (make-coords :pos (float-vector (+ 400 (* 100 (sin cnt))) (* 50 (cos cnt)) 0))
                 #'(lambda ()
                     (send (send (send *robot* :larm :end-coords) :copy-worldcoords)
                           :translate #f(0 0 100) :local))
                 )
           :move-target
           (list (send *robot* :larm :end-coords)
                 (send *robot* :rarm :end-coords)
                 )
           :link-list
           (list (send *robot* :link-list
                       (send (send *robot* :larm :end-coords) :parent)
                       (car (send *robot* :larm :links))
                       )
                 (send *robot* :link-list
                       (send (send *robot* :rarm :end-coords) :parent)
                       (car (send *robot* :rarm :links))
                       )
                 )
           :translation-axis (list :z t)
           :rotation-axis (list nil :z)
           )
     (send *obj* :newcoords (send (send *robot* :larm :end-coords) :copy-worldcoords))
     (send *irtviewer* :draw-objects)
     ))
  )
```

iory, 2018-07-30:
@mmurooka さん
ありがとうございます。
たしかにtarget-coordsを関数としてとらなければ解けない例ですね。

@k-okada 
target-coordsを関数でとる必要性がありそうですが、どうでしょうか？

iory, 2018-08-04:
@k-okada ping

iory, 2018-08-08:
こちらの方にdocumentとdemoのプログラムを追加しました。
https://github.com/euslisp/jskeus/pull/518

表題の件は解決したのでcloseします。

closed 2018-08-08 by iory