euslisp/jskeus #531: irtviewerを作ろうとすると ;; no display is found. exit from (make-irtviewer)

shintarokkk, 2018-12-10:
Ubuntu 16.04に
https://github.com/jsk-ros-pkg/jsk_roseus/blob/master/README.md
の方法でソースからeuslisp等の環境を構築し、hrpsys,hrpsys_ros_bridgeなどを入れて、

$ roseus
1.irteusgl$ (load "package://hrpsys_ros_bridge/euslisp/samplerobot-interface.l")                                                                           
3.irteusgl$ (objects (list (samplerobot)))                                                                                                                 

とすると、

;; (make-irtviewer) executed                                                                                                                               
;; no display is found. exit from (make-irtviewer)                                                                                                         
(#<samplerobot-robot #X64e3588 SampleRobot  0.0 0.0 0.0 / 0.0 0.0 0.0>)

というエラーが出て、irtviewerが表示されません。
ソースを辿ってみて、x::panelが関係あるのかと思って

4.irteusgl$ (instance x::panel :create)                                                                                                                    

としてみたところ、

Call Stack (max depth: 20):                                                                                                                                
  0: at (send #:inst4122 :create)                                                                                                                          
  1: at (send #:inst4122 :create)                                                                                                                          
  2: at (send #:inst4122 :create)                                                                                                                          
  3: at (send #:inst4122 :create)                                                                                                                          
  4: at (let ((#:inst4122 (instantiate x:panel))) (send #:inst4122 :create) #:inst4122)                                                                    
  5: at (instance x:panel :create)                                                                                                                         
  6: at #<compiled-code #X52e8d80>                                                                                                                         
/home/shintarokkk/catkin_ws/devel/share/euslisp/jskeus/eus/Linux64/bin/irteusgl 0 error:  ;; visual-depth: no such visual nil registered. in (send #:inst4122 :create) 

というエラーが出ました。
これ以上はエラー原因が突き止められなそうなので、よろしければどのあたりが間違っていそうか教えていただけないでしょうか。

k-okada, 2018-12-10:
xlogo としてXが動いているか確認してください．
```
$ roseus
1.irteusgl$ (load
"package://hrpsys_ros_bridge/euslisp/samplerobot-interface.l")
3.irteusgl$ (objects (list (samplerobot)))
```
のところで，編集しているみたいですが，入力したコマンド，その全出力を全てコピー＆ペーストするとよいです．そこにメッセージがでているはずです．

shintarokkk, 2018-12-10:
ディスプレイの調子が悪かったのでPCを再起動して再び試したところ、正常にirtviewerが立ち上がるようになっていました...

エラーが出ている時にやればよかったのですが、正常にirtviewerが立ち上がる状態でxlogoを実行すると
[image]
このロゴが表示されました。

closed 2018-12-10 by shintarokkk