euslisp/jskeus #586: body+ ができない

atnarita, 2020-10-26:
ロボットシステムの課題2に取り組んでいるのですが、``body+``をするとエラーが出てしまいます。
何が原因なのでしょうか。

ソースコード
```
(defun make-head nil
  (let (s1 s2 s3 s4)
    (setq s1 (make-cube 260 360 20))
    (setq s2 (make-cube 20 20 20))
    (setq s3 (make-cube 20 20 20))
    (setq s4 (make-cube 200 300 10))
    (send s2 :locate #f(-120.0 -140.0 -30.0))
    (send s3 :locate #f(-120.0 140.0 -30.0))
    (send s4 :locate #f(0.0 0.0 -5.0))
    (setq s1 (body- (body+ s2 (body+ s1 s3)) s4))
    (send s1 :set-color :gray80)
    s1)))

(setq h1 (instance bodyset-link :init (make-cascoords) :bodies (list (make-head))))
(objects (list h1)
```

以下エラー文                                                                           
```                                                                                 
o1.irteusgl$ (load "2.l")                                                                  
Call Stack (max depth: 20):                                                                
  0: at (body+ s1 s3)                                                                      
  1: at (body+ s2 (body+ s1 s3))                                                           
  2: at (body- (body+ s2 (body+ s1 s3)) s4)                                                
  3: at (setq s1 (body- (body+ s2 (body+ s1 s3)) s4))                                      
  4: at (let (s1 s2 s3 s4) (setq s1 (make-cube 260 360 20)) (setq s2 (make-cube 20 20 20))\
 (setq s3 (make-cube 20 20 20)) (setq s4 (make-cube 200 300 10)) (send s2 :locate #f(-120.\
0 -140.0 -30.0)) (send s3 :locate #f(-120.0 140.0 -30.0)) (send s4 :locate #f(0.0 0.0 -5.0\
)) (setq s1 (body- (body+ s2 (body+ s1 s3)) s4)) (send s1 :set-color :gray80) s1)          
  5: at (make-head)                                                                        
  6: at (list (make-head))                                                                 
  7: at (send #:inst63 :init (make-cascoords) :bodies (list (make-head)))                  
  8: at (let ((#:inst63 (instantiate bodyset-link))) (send #:inst63 :init (make-cascoords)\
 :bodies (list (make-head))) #:inst63)                                                     
  9: at (instance bodyset-link :init (make-cascoords) :bodies (list (make-head)))          
  10: at (setq h1 (instance bodyset-link :init (make-cascoords) :bodies (list (make-head))\
))                                                                                         
  11: at (load "2.l")                                                                      
  12: at #<compiled-code #X5606cefa03e8>                                                   
irteusgl 0 error: float vector expected in (body+ s1 s3)                                   
                                                                                           
```

k-okada, 2020-10-26:
基本的には`body+`は重なりのある物体間で呼び出します．
なので，
```
(defun make-head nil
  (let ();s1 s2 s3 s4)                                                                                                    
    (setq s1 (make-cube 260 360 20))
    (setq s2 (make-cube 20 20 40))
    (setq s3 (make-cube 20 20 40))
    (setq s4 (make-cube 200 300 10))
    (send s2 :locate #f(-120.0 -140.0 -30.0))
    (send s3 :locate #f(-120.0 140.0 -30.0))
    (send s4 :locate #f(0.0 0.0 -5.0))
    (setq s1 (body- (body+ s2 (body+ s1 s3)) s4))
    (send s1 :set-color :gray80)
    s1)))

(setq h1 (instance bodyset-link :init (make-cascoords) :bodies (list (make-head))))
(objects (list h1))
```
とすると通ります．
一方で，そもそも重なりがない物体からリンクを作りたければ，
```
(defun make-head2 nil
  (let ();s1 s2 s3 s4)                                                                                                    
    (setq s1 (make-cube 260 360 20))
    (setq s2 (make-cube 20 20 20))
    (setq s3 (make-cube 20 20 20))
    (setq s4 (make-cube 200 300 10))
    (send s2 :locate #f(-120.0 -140.0 -30.0))
    (send s3 :locate #f(-120.0 140.0 -30.0))
    (send s4 :locate #f(0.0 0.0 -5.0))
    (setq s1 (body- s1 s4))
    (send s1 :set-color :gray80)
    (list s1 s2 s3)))

(setq h2 (instance bodyset-link :init (make-cascoords) :bodies (make-head2)))
(objects (list h2))

```
`bodyset` を活用すると良いです．

atnarita, 2020-10-26:
ありがとうございます。

closed 2020-10-26 by atnarita