euslisp/jskeus #587: eusでの座標の矢印をいい感じのサイズ感で目立つように表示させたい

k-okada, 2020-10-31:
@Kanazawanaoaki

eusでの座標の矢印をいい感じのサイズ感で目立つように表示させた画像を作りたいのですが，arrow-object.lだとサイズを大きすぎて上手く伝わらなくて，draw-onで座標を表示すると白くて微妙に見立たなくて伝わらない気がします．

arrowのサイズを変えるか，draw-onの矢印に色をつける方法は無いでしょか？

[image]

k-okada, 2020-10-31:
`(send (make-coords) :draw-on :flush t :size 100 :color #f(1 0 0) :width 5)`

とできます．

[image]

arrow-object は変更できないですね．これの元定義は
https://github.com/jsk-ros-pkg/euslib/blob/9622f7d004473d8a2b48f7764b03986183a20293/demo/chris/jskmapmodel.l#L1475-L1487
にあるので，
https://gist.github.com/k-okada/2fad270cecb6ae52bc5fa1874667e323
みたいにして，
`(progn (load "/tmp/arrow.l")(objects (list (make-cube 100 100 1)  (make-arrow-body 10))))`
みたいにできます．

Kanazawanaoaki, 2020-10-31:
ありがとうございます．解決しました．
[image]
[image]

pazeshun, 2021-11-20:
```
(send (make-coords) :draw-on :flush t :size 100 :color #f(1 0 0) :width 5)
```
だと、マウスで回そうとすると消えてしまいました。
僕が困ったのは、複数の力ベクトルの空間的な分布を確認しようとした時で、
```
(dolist (v view-vecs)
  (send *irtviewer* :viewer :draw-arrow #f(0 0 0) v))
```
で書いたベクトル群を一方向から見るだけだと、本当は別方向のベクトルが重なって見えたりするので大変わかりにくく、マウスで回そうとしたら消えてしまって困りました。
結局、
https://gist.github.com/k-okada/2fad270cecb6ae52bc5fa1874667e323
の`make-single-arrow-body`を使うことで解決しそうです。
余裕ができたらどこかにPRしようと思いました。