euslisp/jskeus #590: face を分割するような hole を設定すると三角メッシュがうまく生成されない

yuki-shark, 2020-12-12:
@Naoki-Hiraoka

`(geometry::face-to-triangle)` を使っていて，三角メッシュに切るところで #585 で修正してくれたバグに悩まされていたので非常に助かりました．
もう一点困っていて， hole が元の face を2分割するようにセットされた場合，三角メッシュがうまく生成出きないのですが対処法など心当たりがあれば教えてもらいたいです．

* うまくいくケース（hole による face の分割なし）
```
(setq f1 (instance face
                   :init
                   :vertices (list #f(-200.0 -200.0 0.0)
                                   #f(-200.0 200.0 0.0)
                                   #f(200.0 200.0 0.0)
                                   #f(200.0 -200.0 0.0))
                   :holes
                   (list (instance hole :init :vertices
                                   (list #f(100.0 100.0 0.0)
                                         #f(-100.0 100.0 0.0)
                                         #f(-100.0 -100.0 0.0)
                                         #f(100.0 -100.0 0.0))))))
(setq tri1 (geometry::face-to-triangle f1))
(dotimes (i (length tri1)) (send (elt tri1 i) :draw-on :flush t))
```
[image]

* うまくいかないケース（hole による face の分割あり）
```
(setq f2 (instance face
                   :init
                   :vertices
                   (list #f(-200.0 -200.0 0.0)
                         #f(-200.0 200.0 0.0)
                         #f(200.0 200.0 0.0)
                         #f(200.0 -200.0 0.0))
                   :holes
                   (list (instance hole :init :vertices
                                   (list #f(300.0 100.0 0.0)
                                         #f(-300.0 100.0 0.0)
                                         #f(-300.0 -100.0 0.0)
                                         #f(300.0 -100.0 0.0))))))
(send f2 :draw-on :flush t)
(send (car (send f2 :holes) ) :draw-on :flush t)

(setq tri2 (geometry::face-to-triangle f2))
```
こちらは tri2 に nil が入ります．
[image]

Naoki-Hiraoka, 2020-12-12:
hole による face の分割があることは、もとのアルゴリズムが想定していないので、できないのではないかと思います。

いったん厚みをつけて直方体にして、holeの部分をbody-で除去して、最後に法線ベクトルが上向きの面それぞれに対して個別に`(geometry::face-to-triangle)`すると計算できるのではないかと思いました。

yuki-shark, 2020-12-12:
直方体にするという発想がなかったです．
それなら出来そうですね．ありがとうございます．

closed 2020-12-12 by yuki-shark