euslisp/jskeus #594: calc-cog-jacobian-from-link-list を使って CoG の変化を行動前に予測したい

hiroya1224, 2020-12-24:
重心位置 CoG が関節角によってどのように変化するかを調べたくて， `:calc-cog-jacobian-from-link-list` を利用しようと思い検証しています．
そこで以下のようにして，ヤコビ行列から求めた重心位置と実際の重心位置を比較することを試してみました:
```
(let* ( ;; save init state
        (init-av (send *robot* :angle-vector))
        (init-cog (copy-seq (send *robot* :centroid)))
        ;; calc jacobian matrix
        (jm-cog-q (scale-matrix 1e3 (send *robot* :calc-cog-jacobian-from-link-list))) ;; scale [m] -> [mm]
        ;; set delta-q
        (delta-q (make-sequence float-vector (length init-av) :initial-element 0.01)) ;; 0.01 is arbitrary
        ;; estimation of delta-cog
        (J-deltaq-at-q (transform jm-cog-q delta-q))
        jm-cog-q-plus-deltaq J-deltaq-at-q-plus-deltaq)
    
    ;; send an angle vector to the *robot*
    (send *robot* :angle-vector (v+ init-av delta-q))
    
    (setq jm-cog-q-plus-deltaq (scale-matrix 1e3 (send *robot* :calc-cog-jacobian-from-link-list)))
    (setq J-deltaq-at-q-plus-deltaq (transform jm-cog-q-plus-deltaq delta-q))
      
    ;; calc actual delta-cog and show result
    (warn "estimated-cog: ~A ~~ ~A,~%actual-cog: ~A.~%" 
        (v+ init-cog J-deltaq-at-q) (v+ init-cog J-deltaq-at-q-plus-deltaq)
        (send *robot* :centroid))
)
```
以上を試してみると，たとえば
```
estimated-cog: #f(6.63863 3.34355 44.82) ~ #f(6.63852 3.3434 44.8203),
actual-cog: #f(7.07784 2.83891 44.8012).
```
という結果が得られました．ここで estimated-cog に表示された2つの結果は，q=q_0 のときと q=q_0+Δq のときの ∂CoG/∂q ・Δqの値です．
Taylor の定理によれば(諸々の仮定は満たしているとする)， ある 0<θ<1 が存在して，
![\begin{equation*}
\mathrm{CoG}({\boldsymbol q}_0 + \Delta {\boldsymbol q}) = \left. \mathrm{CoG}({\boldsymbol q}_0) + \frac{\partial{\mathrm{CoG}}}{\partial{\boldsymbol q}} \right| _{{\boldsymbol q} = {\boldsymbol q}_0 + \theta \Delta {\boldsymbol q}} \Delta {\boldsymbol q}
\end{equation*}
](https://render.githubusercontent.com/render/math?math=%5Clarge+%5Cdisplaystyle+%5Cbegin%7Bequation%2A%7D%0A%5Cmathrm%7BCoG%7D%28%7B%5Cboldsymbol+q%7D_0+%2B+%5CDelta+%7B%5Cboldsymbol+q%7D%29+%3D+%5Cleft.+%5Cmathrm%7BCoG%7D%28%7B%5Cboldsymbol+q%7D_0%29+%2B+%5Cfrac%7B%5Cpartial%7B%5Cmathrm%7BCoG%7D%7D%7D%7B%5Cpartial%7B%5Cboldsymbol+q%7D%7D+%5Cright%7C+_%7B%7B%5Cboldsymbol+q%7D+%3D+%7B%5Cboldsymbol+q%7D_0+%2B+%5Ctheta+%5CDelta+%7B%5Cboldsymbol+q%7D%7D+%5CDelta+%7B%5Cboldsymbol+q%7D%0A%5Cend%7Bequation%2A%7D%0A)
となるようですが， actual-cog は estimated-cog の範囲からはみ出しており，不適っぽい気がします
(正確には単調変化するとは限らないので途中も見るべきですが，望み薄そうなので省略)．
この現象は Δq の幅を0.01から変化させたり，姿勢を変化させても起き続けます． 

どこか勘違いしているところがあるでしょうか？`:calc-cog-jacobian-from-link-list` で得られるのは ∂CoG/∂q のことだと思っていたのですが，違いますでしょうか．

@k-okada @Naoki-Hiraoka @haraduka

YoheiKakiuchi, 2020-12-25:
コード動かさないで返答するけど、ヤコビに掛ける角度はラジアンじゃないかな。
それで、angle-vectorはdegree

hiroya1224, 2020-12-25:
垣内先生，ありがとうございます．なるほど，完全に盲点でした．．．
上記コードで
```
    ;; send an angle vector to the *robot*
    (send *robot* :angle-vector (v+ init-av delta-q))
```
の部分を，
```
    ;; send an angle vector to the *robot*
    (send *robot* :angle-vector (v+ init-av (scale (/ 180.0 PI) delta-q)))
```
と修正して再度試行したところ，結果は
```
estimated-cog: #f(-2.97564 11.5463 48.8722) ~ #f(-2.9663 11.5354 48.8823),
actual-cog: #f(-2.97102 11.5408 48.8773).
```
のようになり，無事範囲内におさまりました．ありがとうございます．

closed 2020-12-25 by hiroya1224