euslisp/jskeus #601: いろいろなロボットの eusmodel を一括で表示させる方法を知りたい

hiroya1224, 2021-05-17:
#474 のように色々なロボットを表示させようとしたところ，

```
8: at (load "package://hrpsys_ros_bridge_tutorials/euslisp/hrp2jsknts-interface.l")
/home/leus/jskeus/eus/Linux64/bin/irteusgl: ERROR th=0  file #P"/home/leus/catkin_ws/src/rtm-ros-r\
obotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/hrp2jsknts.l" not found in (apply #'ro\
s::load-org-for-ros ros::fullname args)E: 
```
と止まってしまいます． ls してみると
```
$ ls /home/leus/catkin_ws/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/
CHIDORI.PDgains.sav                    gr001.yaml
CHIDORI.PDgains_sim.dat                hand_force_calib_offset_HRP2JSKNT
HIRONX_controller_config.yaml          hand_force_calib_offset_HRP2JSKNTS
HRP2G.urdf.xacro                       hand_force_calib_offset_JAXON_RED
HRP2JSK.PDgain.dat                     hand_force_calib_offset_THK003_CLOSE_JAXON
HRP2JSK.urdf.xacro                     hand_force_calib_offset_thumb_60deg_HRP2JSK
HRP2JSKNT.PDgain.dat                   hironxjsk.yaml
HRP2JSKNT.urdf.xacro                   hrp2g.yaml
HRP2JSKNTS.PDgain.dat                  hrp2jsk.yaml
HRP2JSKNTS.urdf.xacro                  hrp2jsknt.yaml
HRP2JSKNTS_body_customHEAD.urdf        hrp2jsknt_with_3hand.yaml
HRP2JSKNTS_bush_customizer_param.conf  hrp2jsknts.yaml
HRP2JSKNTS_customHEAD.urdf             hrp2jsknts_with_3hand.yaml
HRP2JSKNTS_meshes                      hrp2w.yaml
HRP2JSKNT_bush_customizer_param.conf   hrp3hand_l.yaml
HRP2JSK_bush_customizer_param.conf     hrp3hand_r.yaml
HRP2W.urdf.xacro                       hrp4c.yaml
HRP4C_controller_config.yaml           hrp4r.yaml
JAXON.urdf.xacro                       jaxon.yaml
JAXON_BLUE.PDgains.sav                 jaxon_blue.yaml
JAXON_BLUE.PDgains_sim.dat             jaxon_red.yaml
JAXON_RED.urdf.xacro                   kawada-hironx.yaml
JAXON_RED_controller_config.yaml       pa10.yaml
PDgains.sav                            robotiq_l.urdf.xacro
STARO.urdf.xacro                       robotiq_r.urdf.xacro
SampleRobot_controller_config.yaml     samplerobot.yaml
TESTMDOFARM.wrl                        staro.yaml
chidori.yaml                           testmdofarm.yaml
darkgoldenrod-stairs.wrl               tqleg0.yaml
darwin.yaml                            urataleg.yaml
gen_hand_attached_staro_model.sh       yaskawasmartpal5.yaml
gen_sensor_attached_staro_model.sh     ystleg.yaml
```
となり， .l ファイルが見当たらないようです．

```
catkin build rtmros_common
```
は行いました．対処法を教えてください．

Naoki-Hiraoka, 2021-05-17:
https://github.com/start-jsk/rtmros_hrp2 のREADMEの指示に従って環境構築してください。

hiroya1224, 2021-05-17:
rtmros_hrp2 が 404 で見えない．．．
鍵リポジトリが見えないようなので，パーミッションの更新をお願いしたいです．
@k-okada @YoheiKakiuchi

hiroya1224, 2021-05-17:
パーミッションの更新ありがとうございます．
README の指示に従って進めたところロボットモデルが入り，利用できました．
ありがとうございます．

closed 2021-05-17 by hiroya1224