euslisp/jskeus #634: min-max-tableのcitationはありますか

Naoki-Hiraoka, 2023-06-18:
論文等でmin-max-tableに触れなければならない場合に、citationするべき論文はあるでしょうか.

自分が遡れた一番古い記述はhttps://sourceforge.net/p/jskeus/tickets/43/ ですが、さらにその前のrbrain時代からmin-max-tableはあったようで、それ以上遡る方法が分かりませんでした。

k-okada, 2023-06-18:
https://ieeexplore.ieee.org/stamp/stamp.jsp?tp=&arnumber=4059025
かな。最初にEusで実装したか、hrpsysで実装したか忘れてしまったけど、hrpsysで安全装置としてつかって、安全装置に引っかからないようにEusで実装して動作生成時に活用したのか
あるいは逆でEusで実装したものをhrpsysコードに自動生成していたか、fig5のコードは何かそういう感じもします
--
◉ Kei Okada

Naoki-Hiraoka, 2023-06-19:
ありがとうございます。

hrpsysにmin-max-tableが入ったのは2014年なので(https://github.com/fkanehiro/hrpsys-base/pull/433 )、最初にeusで実装してeusの動作生成時に活用していて、その後eusで計算したtableをhrpsysで読み込んで安全装置としても使うようにしたのだと思っていましたが、Fig. 5のコードはそれより前のhrpsysのプラグインで使われていた感じもします。

closed 2023-06-19 by Naoki-Hiraoka