euslisp/jskeus #639: EuslispのGUI画面が表示されない

HisaakiIida, 2023-11-18:
Euslispを家のPCで起動しようとしたところ、下記エラーメッセージのあとに(full-body-ik)などの一覧が出力されました。

configuring by "/opt/ros/noetic/share/euslisp/jskeus/eus//lib/eusrt.l"
;; readmacro ;; object ;; packsym ;; common ;; constants ;; stream ;; string ;; loader ;; pprint ;; process ;; hashtab ;; array ;; mathtran ;; eusdebug ;; eusforeign ;; extnum ;; coordinates ;; tty ;; history ;; toplevel ;; trans ;; comp ;; builtins ;; par ;; intersection ;; geoclasses ;; geopack ;; geobody ;; primt ;; compose ;; polygon ;; viewing ;; viewport ;; viewsurface ;; hid ;; shadow ;; bodyrel ;; dda ;; helpsub ;; eushelp ;; xforeign ;; Xdecl ;; Xgraphics ;; Xcolor ;; Xeus ;; Xevent ;; Xpanel ;; Xitem ;; Xtext ;; Xmenu ;; Xscroll ;; Xcanvas ;; Xtop ;; Xapplwin 
No protocol specified
Xserver connection failed;; pixword ;; RGBHLS ;; convolve ;; piximage ;; pbmfile ;; image_correlation ;; oglforeign ;; gldecl ;; glconst ;; glforeign ;; gluconst ;; gluforeign ;; glxconst ;; glxforeign ;; eglforeign ;; eglfunc ;; glutil ;; gltexture ;; glprim ;; gleus ;; glview ;; toiv-undefined ;; fstringdouble irtmath irtutil irtc irtgeoc irtgraph gnuplotlib ___time ___pgsql irtgeo euspqp pqp irtscene irtmodel irtdyna irtrobot irtsensor irtbvh irtcollada irtstl irtwrl irtpointcloud eusbullet bullet irtcollision irtx eusjpeg euspng png irtimage irtglrgb 
;; extending gcstack 0x5598df2bf680[16374] --> 0x5598df71abf0[32748] top=3d4b
irtgl irtglc irtviewer 
EusLisp 9.29( 1.2.5) for Linux64 created on ip-10-0-1-5(Tue Jun 20 10:34:21 PST 2023)
;; no display is found. exit from (make-irtviewer)
;; no display is found. exit from (make-irtviewer)

その状態でfull-body-ikを試してみると、

loop:   0
union-link-list: (:torso-link0 :torso-link1 :larm-link0 :larm-link1 :larm-link2 :larm-link3 :larm-link4 :larm-link5 :larm-link6)
move-target: (#<cascaded-coords #X5598df842678  -9.936 -17.568 1376.588 / 1.055 -1.409 -1.606>)
targe-coordst: (#<coordinates #X5598df335428  0.0 0.0 1500.0 / 0.0 0.0 0.0>)
vel-pos :100.000   0.000   0.025 
vel-rot :
vel-pos-norm : 100.000/  1.000
vel-rot-norm :   0.000/  0.017
collision-links : (:waist :torso-link1 :larm-link3 :rarm-link3)
parnet-links    : (nil :waist :larm-link2 :rarm-link2)
valid-link-list : (:torso-link1 :larm-link0 :larm-link1 :larm-link2 :larm-link3 :larm-link4 :larm-link5 :larm-link6)
index-links     : (nil 0 4 nil)
angle :  -33.1   -14.6  -150.0    23.9    -1.3   -13.3     4.3     0.1     1.1 
 min  :  -45.0   -90.0  -150.0   -30.0   -90.0  -180.0   -90.0   -90.0   -90.0 
 max  :   45.0    90.0   150.0   180.0    90.0     0.0    90.0    90.0    90.0 
usrwei:  1.000   1.000   1.000   1.000   1.000   1.000   1.000   1.000   1.000 
ocost :  8.893   0.218 4297.135   0.911   0.018  14.492   0.062   0.001   0.016 
cost  :  8.893   0.218 4297.135   0.911   0.018  14.493   0.062   0.001   0.016 
weight:  0.101   1.000   0.000   0.523   0.982   0.065   1.000   1.000   1.000 
k     :  0.031 (manipulability:  0.021, gain:  0.050, limit:  0.100, len:9)
col-distance : 129.799(<200.000) #f(  0.285   0.685   0.671 ) (:torso-link1 :rarm-link3)
col-distance : 233.029(<200.000) #f( -0.252  -0.240  -0.938 ) (:torso-link1 :larm-link3)
col-distance : 436.465(<200.000) #f(  0.095   0.108   0.990 ) (:larm-link3 :waist)
col-distance : 647.365(<200.000) #f(  0.298   0.386   0.873 ) (:larm-link3 :rarm-link3)
coljnt: -0.000   0.001   0.000   0.000   0.000   0.000   0.000   0.000   0.000 
colnul: -0.000   0.001   0.000   0.000   0.000   0.000   0.000   0.000   0.000 
colvel: -0.000   0.001   0.000   0.000   0.000   0.000   0.000   0.000   0.000 
blend :  0.390
nspace:  0.031   0.001   0.000   0.071   0.000  -0.027  -0.001  -0.000  -0.000 
J     : -0.003  -0.000  -0.050  -0.001  -0.000   0.004  -0.000   0.000  -0.000 
         0.042  -0.026   0.003   0.513   0.055   0.018  -0.001   0.050   0.000 
         0.356  -0.649  -0.471   0.027   0.004  -0.244  -0.000  -0.000  -0.050 
d(JJt):  0.000
J#t   : -0.008  -0.007  -0.000  -0.004   0.003   0.008  -0.000   0.003  -0.000 
         0.013   0.054   0.000   1.538   0.311   0.012  -0.005   0.287   0.015 
         0.075  -1.377  -0.000  -0.052  -0.008  -0.034   0.000  -0.015  -0.107 
x     :  0.100   0.000   0.000 
J#x   : -0.048  -0.042  -0.002  -0.024   0.018   0.044  -0.000   0.017  -0.002 
aspeed: -0.000   0.001   0.000   0.000   0.000   0.000   0.000   0.000   0.000 
J#x   : -0.029  -0.025  -0.001  -0.015   0.011   0.027  -0.000   0.010  -0.001 
Ny    :  0.029   0.025   0.000   0.015  -0.011  -0.027  -0.001  -0.010   0.001 
dav   :  0.000  -0.000  -0.001   0.000  -0.000   0.000  -0.001  -0.000  -0.000 
dav^  :  0.000  -0.000  -0.001   0.000  -0.000   0.000  -0.001  -0.000  -0.000 
;; (window-main-one nil) was called, but no display is found.

というエラーメッセージが繰り返し出力され、GUIの画面が表示されません。
いろいろと調べてX11やDISPLAY環境変数の変更なども試してみましたが、改善するようすがないので質問させていただきます。

k-okada, 2023-11-19:
環境は何を使っていますか？ linux? windows + wsl ? mac + docker ? ???

HisaakiIida, 2023-11-19:
linuxで、学科PCと同じubuntu 20.04を使っています。

k-okada, 2023-11-19:
```
sudo apt install -y mesa-utils
DISPLAY=:0.0 glxinfo
```
の結果をすべて貼り付けてください。

HisaakiIida, 2023-11-19:
Invalid MIT-MAGIC-COOKIE-1 keyError: unable to open display :0.0
と出ました。

k-okada, 2023-11-19:
この時の画面全体のキャプチャを張ってみてください。
--
◉ Kei Okada

HisaakiIida, 2023-11-19:
[image]
こちらです

k-okada, 2023-11-19:
xhost +
としてから、もう一度 DISPLAY=:0.0 glxinfo  をしてみてください。
--
◉ Kei Okada

HisaakiIida, 2023-11-19:
変わらずInvalid MIT-MAGIC-COOKIE-1 keyError: unable to open display :0.0が表示されます。

k-okada, 2023-11-19:
xhost としたときの画面のキャプチャを送ってください
--
◉ Kei Okada

HisaakiIida, 2023-11-19:
access control disabled, clients can connect from any host
とだけ表示されています。

k-okada, 2023-11-19:
cat /etc/hosts
hostname
の出力をキャプチャして送ってください

HisaakiIida, 2023-11-19:
[image]
こちらです

k-okada, 2023-11-19:
問題ないように見えますね。
cat /vat/log/Xorg.0.log
と
env
の出力もすべて送ってください。

HisaakiIida, 2023-11-19:
[image]

$ env
SHELL=/bin/bash
ROS_VERSION=1
SESSION_MANAGER=local/kokumapc:@/tmp/.ICE-unix/1992,unix/kokumapc:/tmp/.ICE-unix/1992
QT_ACCESSIBILITY=1
COLORTERM=truecolor
XDG_CONFIG_DIRS=/etc/xdg/xdg-ubuntu:/etc/xdg
XDG_MENU_PREFIX=gnome-
GNOME_DESKTOP_SESSION_ID=this-is-deprecated
PKG_CONFIG_PATH=/opt/ros/noetic/lib/pkgconfig:/opt/ros/noetic/lib/x86_64-linux-gnu/pkgconfig
ROS_PYTHON_VERSION=3
GNOME_SHELL_SESSION_MODE=ubuntu
SSH_AUTH_SOCK=/run/user/1000/keyring/ssh
XMODIFIERS=@im=ibus
DESKTOP_SESSION=ubuntu
SSH_AGENT_PID=1890
ROS_PACKAGE_PATH=/opt/ros/noetic/share
GTK_MODULES=gail:atk-bridge
ROSLISP_PACKAGE_DIRECTORIES=
PWD=/home/hisaakiiida
LOGNAME=hisaakiiida
XDG_SESSION_DESKTOP=ubuntu
XDG_SESSION_TYPE=x11
GPG_AGENT_INFO=/run/user/1000/gnupg/S.gpg-agent:0:1
XAUTHORITY=/run/user/1000/gdm/Xauthority
GJS_DEBUG_TOPICS=JS ERROR;JS LOG
WINDOWPATH=2
HOME=/home/hisaakiiida
USERNAME=hisaakiiida
IM_CONFIG_PHASE=1
LANG=ja_JP.UTF-8
ROS_ETC_DIR=/opt/ros/noetic/etc/ros
LS_COLORS=rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.zst=01;31:*.tzst=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.wim=01;31:*.swm=01;31:*.dwm=01;31:*.esd=01;31:*.jpg=01;35:*.jpeg=01;35:*.mjpg=01;35:*.mjpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:
XDG_CURRENT_DESKTOP=ubuntu:GNOME
VTE_VERSION=6003
CMAKE_PREFIX_PATH=/opt/ros/noetic
GNOME_TERMINAL_SCREEN=/org/gnome/Terminal/screen/34a53313_cf88_49d5_99a8_a704097867a0
INVOCATION_ID=729a5ab6255b4dc680ab520b8e28f68b
MANAGERPID=1711
GJS_DEBUG_OUTPUT=stderr
LESSCLOSE=/usr/bin/lesspipe %s %s
XDG_SESSION_CLASS=user
PYTHONPATH=/opt/ros/noetic/lib/python3/dist-packages
TERM=xterm-256color
LESSOPEN=| /usr/bin/lesspipe %s
USER=hisaakiiida
GNOME_TERMINAL_SERVICE=:1.128
DISPLAY=:1
SHLVL=1
ROS_MASTER_URI=http://localhost:11311
QT_IM_MODULE=ibus
LD_LIBRARY_PATH=/opt/ros/noetic/share/euslisp/jskeus/eus//Linux64/lib:/opt/ros/noetic/lib:/opt/ros/noetic/lib/x86_64-linux-gnu
XDG_RUNTIME_DIR=/run/user/1000
ARCHDIR=Linux64
EUSDIR=/opt/ros/noetic/share/euslisp/jskeus/eus/
JOURNAL_STREAM=8:47903
XDG_DATA_DIRS=/usr/share/ubuntu:/usr/local/share/:/usr/share/:/var/lib/snapd/desktop
PATH=/opt/ros/noetic/share/euslisp/jskeus/eus//Linux64/bin:/opt/ros/noetic/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
GDMSESSION=ubuntu
DBUS_SESSION_BUS_ADDRESS=unix:path=/run/user/1000/bus
ROS_ROOT=/opt/ros/noetic/share/ros
ROS_DISTRO=noetic
_=/usr/bin/env

k-okada, 2023-11-19:
glxinfo
だけをするとどうなりますか？

HisaakiIida, 2023-11-19:
~$ glxinfo
name of display: :1
display: :1  screen: 0
direct rendering: Yes
server glx vendor string: SGI
server glx version string: 1.4
server glx extensions:
    GLX_ARB_context_flush_control, GLX_ARB_create_context, 
    GLX_ARB_create_context_no_error, GLX_ARB_create_context_profile, 
    GLX_ARB_fbconfig_float, GLX_ARB_framebuffer_sRGB, GLX_ARB_multisample, 
    GLX_EXT_create_context_es2_profile, GLX_EXT_create_context_es_profile, 
    GLX_EXT_fbconfig_packed_float, GLX_EXT_framebuffer_sRGB, 
    GLX_EXT_import_context, GLX_EXT_libglvnd, GLX_EXT_no_config_context, 
    GLX_EXT_texture_from_pixmap, GLX_EXT_visual_info, GLX_EXT_visual_rating, 
    GLX_MESA_copy_sub_buffer, GLX_OML_swap_method, GLX_SGIS_multisample, 
    GLX_SGIX_fbconfig, GLX_SGIX_pbuffer, GLX_SGIX_visual_select_group, 
    GLX_SGI_make_current_read
client glx vendor string: Mesa Project and SGI
client glx version string: 1.4
client glx extensions:
    GLX_ARB_context_flush_control, GLX_ARB_create_context, 
    GLX_ARB_create_context_no_error, GLX_ARB_create_context_profile, 
    GLX_ARB_create_context_robustness, GLX_ARB_fbconfig_float, 
    GLX_ARB_framebuffer_sRGB, GLX_ARB_get_proc_address, GLX_ARB_multisample, 
    GLX_ATI_pixel_format_float, GLX_EXT_buffer_age, 
    GLX_EXT_create_context_es2_profile, GLX_EXT_create_context_es_profile, 
    GLX_EXT_fbconfig_packed_float, GLX_EXT_framebuffer_sRGB, 
    GLX_EXT_import_context, GLX_EXT_no_config_context, GLX_EXT_swap_control, 
    GLX_EXT_swap_control_tear, GLX_EXT_texture_from_pixmap, 
    GLX_EXT_visual_info, GLX_EXT_visual_rating, GLX_INTEL_swap_event, 
    GLX_MESA_copy_sub_buffer, GLX_MESA_multithread_makecurrent, 
    GLX_MESA_query_renderer, GLX_MESA_swap_control, GLX_NV_float_buffer, 
    GLX_OML_swap_method, GLX_OML_sync_control, GLX_SGIS_multisample, 
    GLX_SGIX_fbconfig, GLX_SGIX_pbuffer, GLX_SGIX_visual_select_group, 
    GLX_SGI_make_current_read, GLX_SGI_swap_control, GLX_SGI_video_sync
GLX version: 1.4
GLX extensions:
    GLX_ARB_context_flush_control, GLX_ARB_create_context, 
    GLX_ARB_create_context_no_error, GLX_ARB_create_context_profile, 
    GLX_ARB_fbconfig_float, GLX_ARB_framebuffer_sRGB, 
    GLX_ARB_get_proc_address, GLX_ARB_multisample, 
    GLX_EXT_create_context_es2_profile, GLX_EXT_create_context_es_profile, 
    GLX_EXT_fbconfig_packed_float, GLX_EXT_framebuffer_sRGB, 
    GLX_EXT_import_context, GLX_EXT_no_config_context, 
    GLX_EXT_texture_from_pixmap, GLX_EXT_visual_info, GLX_EXT_visual_rating, 
    GLX_MESA_copy_sub_buffer, GLX_MESA_query_renderer, GLX_OML_swap_method, 
    GLX_SGIS_multisample, GLX_SGIX_fbconfig, GLX_SGIX_pbuffer, 
    GLX_SGIX_visual_select_group, GLX_SGI_make_current_read
Extended renderer info (GLX_MESA_query_renderer):
    Vendor: Mesa/X.org (0xffffffff)
    Device: llvmpipe (LLVM 12.0.0, 256 bits) (0xffffffff)
    Version: 21.2.6
    Accelerated: no
    Video memory: 31830MB
    Unified memory: no
    Preferred profile: core (0x1)
    Max core profile version: 4.5
    Max compat profile version: 3.1
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.2
OpenGL vendor string: Mesa/X.org
OpenGL renderer string: llvmpipe (LLVM 12.0.0, 256 bits)
OpenGL core profile version string: 4.5 (Core Profile) Mesa 21.2.6
OpenGL core profile shading language version string: 4.50
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
    GL_3DFX_texture_compression_FXT1, GL_AMD_conservative_depth, 
    GL_AMD_draw_buffers_blend, GL_AMD_gpu_shader_int64, 
    GL_AMD_multi_draw_indirect, GL_AMD_query_buffer_object, 
[... 268 lines cut ...]
    GL_OES_packed_depth_stencil, GL_OES_primitive_bounding_box, 
    GL_OES_required_internalformat, GL_OES_rgb8_rgba8, GL_OES_sample_shading, 
    GL_OES_sample_variables, GL_OES_shader_image_atomic, 
    GL_OES_shader_io_blocks, GL_OES_shader_multisample_interpolation, 
    GL_OES_standard_derivatives, GL_OES_stencil8, GL_OES_surfaceless_context, 
    GL_OES_tessellation_point_size, GL_OES_tessellation_shader, 
    GL_OES_texture_3D, GL_OES_texture_border_clamp, GL_OES_texture_buffer, 
    GL_OES_texture_cube_map_array, GL_OES_texture_float, 
    GL_OES_texture_float_linear, GL_OES_texture_half_float, 
    GL_OES_texture_half_float_linear, GL_OES_texture_npot, 
    GL_OES_texture_stencil8, GL_OES_texture_storage_multisample_2d_array, 
    GL_OES_texture_view, GL_OES_vertex_array_object, GL_OES_vertex_half_float, 
    GL_OES_viewport_array

390 GLX Visuals
    visual  x   bf lv rg d st  colorbuffer  sr ax dp st accumbuffer  ms  cav
  id dep cl sp  sz l  ci b ro  r  g  b  a F gb bf th cl  r  g  b  a ns b eat
----------------------------------------------------------------------------
0x021 24 tc  0  32  0 r  y .   8  8  8  8 .  .  0 24  8  0  0  0  0  0 0 None
0x38a 24 tc  0  32  0 r  . .   8  8  8  8 .  .  0  0  0  0  0  0  0  0 0 None
0x38b 24 tc  0  32  0 r  . .   8  8  8  8 .  .  0  0  0 16 16 16 16  0 0 Slow
　：
　：

以降ほぼ同じ構文の繰り返しです。

k-okada, 2023-11-19:
これで irteusgl とすると動くとおもいます．

HisaakiIida, 2023-11-19:
$ irteusgl
configuring by "/opt/ros/noetic/share/euslisp/jskeus/eus//lib/eusrt.l"
;; readmacro ;; object ;; packsym ;; common ;; constants ;; stream ;; string ;; loader ;; pprint ;; process ;; hashtab ;; array ;; mathtran ;; eusdebug ;; eusforeign ;; extnum ;; coordinates ;; tty ;; history ;; toplevel ;; trans ;; comp ;; builtins ;; par ;; intersection ;; geoclasses ;; geopack ;; geobody ;; primt ;; compose ;; polygon ;; viewing ;; viewport ;; viewsurface ;; hid ;; shadow ;; bodyrel ;; dda ;; helpsub ;; eushelp ;; xforeign ;; Xdecl ;; Xgraphics ;; Xcolor ;; Xeus ;; Xevent ;; Xpanel ;; Xitem ;; Xtext ;; Xmenu ;; Xscroll ;; Xcanvas ;; Xtop ;; Xapplwin 
connected to Xserver DISPLAY=:1
X events are being asynchronously monitored.
;; pixword ;; RGBHLS ;; convolve ;; piximage ;; pbmfile ;; image_correlation ;; oglforeign ;; gldecl ;; glconst ;; glforeign ;; gluconst ;; gluforeign ;; glxconst ;; glxforeign ;; eglforeign ;; eglfunc ;; glutil ;; gltexture ;; glprim ;; gleus ;; glview ;; toiv-undefined ;; fstringdouble irtmath irtutil irtc irtgeoc irtgraph gnuplotlib ___time ___pgsql irtgeo euspqp pqp irtscene irtmodel irtdyna irtrobot irtsensor irtbvh irtcollada irtstl irtwrl irtpointcloud eusbullet bullet irtcollision irtx eusjpeg euspng png irtimage irtglrgb 
;; extending gcstack 0x55dbce070680[16374] --> 0x55dbce4fbba0[32748] top=3d4b
irtgl irtglc irtviewer 
EusLisp 9.29( 1.2.5) for Linux64 created on ip-10-0-1-5(Tue Jun 20 10:34:21 PST 2023)
1.irteusgl$ 

となり、エラーは出なくなりましたが相変わらずGUIが出ません。

k-okada, 2023-11-19:
guiはirteusglを立ち上げた後に
```
(load "irteus/demo/demo.l")
(full-body-ik)
```
などとします

HisaakiIida, 2023-11-19:
忘れてました！GUI出ました
ありがとうございました！！